mnesarco/MarzWorkbench #1: Consider renaming the workbench?

luzpaz, 2020-04-05:
It just has better recognition of what it does. We have workbenches like [Flamingo](https://wiki.freecadweb.org/Flamingo_Workbench) and Dodo (which is the Python3/Qt5 port of flamingo) and there is a sentiment that the name of these workbenches don't lend themselves to the workbenches being utilized. So it's something you may want to consider... ?

mnesarco, 2020-04-05:
Something like Marz Guitar Designer?

luzpaz, 2020-04-06:
Seems to work. Lets ask yorik to weigh in.  
CC @yorikvanhavre

yorikvanhavre, 2020-04-06:
Ultimately I think the decision is that author's. Both are fine. Having a more "technical" name such as "guitar and bass designer tool" is of course easier on the user at first read, but "Marz" makes it nicer to "package" into a nice "branding" or "product". Depends much on what your idea is for the future I would say, if you are just releasing a utility or if you are growing it into something to advertise around with pride ;)

In any case the best thing to do IMHO is to have a nice README (which you have!! Impressive!) that starts with a description of the WB, so it appears nicely in the FreeCAD addon manager and user don't loose hours trying to figure out what it does.

mnesarco, 2020-04-06:
@luzpaz @yorikvanhavre 

I have updated the UI and README to reflect the new Name.

[image]

closed 2020-04-11 by mnesarco

luzpaz, 2020-04-12:
I also mean renaming the github repo, not just the workbench name.

mnesarco, 2020-04-12:
Hi @luzpaz , I don't think it is a good idea to rename the Github Repo after I have shared links to it, people will be lost.