mnesarco/MarzWorkbench #3: Headstock transition onto neck

robles1999, 2020-08-06:
Hello Frank.  I am really thankful for your work on this FreeCAD workbench.  I really want to take full advantage of the functionality it has.  I've looked everywhere to see if I find more information on the parameters necessary on the Inkscape file so that everything imports the way it should.  At the moment I don't have any issues with the body import, but the headstock import is not working completely like on you example headstock file.  I am not sure if it's something I am doing wrong or if there is truly a bug in the code.  I am attaching the Inkscape and FreeCAD files for you to review.

I would love to talk to you if at all possible. You can contact me at:
407.733.1841
robles1999@yahoo.com

[image]

[Headstock Issue.zip](https://github.com/mnesarco/MarzWorkbench/files/5035176/Headstock.Issue.zip)

Thank you,
Luis Robles

mnesarco, 2020-08-06:
Hi @robles1999 
Yes there was a bug (a tolerance problem). Fixed now.
BTW, I am rewriting everything from scratch to support better geometry, specially transitions.

The new version will have much better geometry:
[image]
[image]

robles1999, 2020-08-06:
Great. Thank you. 

I think the ability to do the design on Inkscape and imported into FreeCAD to build the 3d was a genius idea!!

Is the correction implemented on the new re-write  or is there an update to the current one?

Also, when do you think the new version will be available?

I would like to propose some other functionalities.
1. Back and top contours 
2. Archtop

Thank you.

robles1999, 2020-08-06:
By the way the new geometry/transition looks great.

mnesarco, 2020-08-06:
@robles1999 

-  The fix is in the current one.
-  I already started the archtop work in the new version.
- Back and Top contours are very easy to implement. Great idea.
- The new version will be public in about one month I hope. It is a complete rewrite, and contains double asymmetric neck profiles, BuiltIn neck profile editor, Custom volute geometry (imported from svg), Custom Heel geometry (svg), way better transition geometry, nut part witth equal center and equal gap support, ... And 1000x performance boost.

[image]

[image]

[image]

robles1999, 2020-08-06:
Wow!!! that's going to be amazing.  

I downloaded the 0.21 version just now to give it a try, and I am still getting the same result on the volute when I add the headstock.

[image]

mnesarco, 2020-08-06:
Hi, mm it is fixed in github but not in 0.21. Give some minutes to upload 0.22

mnesarco, 2020-08-06:
It is fixed in release 0.0.23 now.

robles1999, 2020-08-06:
Ok. Thank you.

By the way I know you are developing and testing in Linux, I have Windows and Mac machines, so I'd be more than happy to help you test the workbench on the 2 systems.

mnesarco, 2020-08-06:
> Ok. Thank you.
> 
> By the way I know you are developing and testing in Linux, I have Windows and Mac machines, so I'd be more than happy to help you test the workbench on the 2 systems.

Thank you, it sounds great. Testing in windows is a pain for me because I use a very slow Windows virtual machine. And testing in Mac is just impossible to me because I don't have one :D

robles1999, 2020-08-06:
For sure. Is the least I can do.  

I am getting to test the .23 release now.

robles1999, 2020-08-06:
The issue is gone.  There's a small issued with the Profile "D".  I don't know if it is that I have to change the parameter when changing the Profiles or is a small code fix.  I don't 100% understand all the parameters yet and I am still playing with them to see the effects of each.  However, here are some pictures of the different profiles and their settings.

[image]
[image]
[image]
[image]

robles1999, 2020-08-06:
The addition of the Constructions lines is great. Another helpful functions would be to be able to Chamfer and Fillet.  I was able to do it by copying the body over to the Parts Design workbench to Fillet and do some boolean functions, but it would be nice to have it in your workbench or have it be a Body rather than a Part.

mnesarco, 2020-08-06:
All the code for transitions (heel and headstock) was rewritten in the new version, I will not port that to this version. :(
Also the neck profile algorithm is completely different in new version and will be not ported back.
Filled and Chamfer are standard FreeCAD features, the idea is to use them with the generated solids.

robles1999, 2020-08-06:
I agree. No need to port it back. 

Thank you for the fix of the volute. This last version alone is of immense help to us non 3D Designer luthiers. It surely has been a struggle to dive into the new technology but I  know it'll be worth it. Again thank you for your work. I look forward to the new version. If you need any help testing the new version please feel free to reach out to me.

robles1999, 2020-08-24:
That looks fantastic!!. Patiently waiting for the complete version.

mnesarco, 2020-08-31:
Advanced custom headstock shape with custom volute, plus double asymmetric neck profile, plus break angle:

[image]

Everything from a simple SVG contour.

Comming soon...

robles1999, 2020-09-01:
Mi hrno. eso esta de maravilla. Is this going to be a stand alone app or a FreeCad workbench?

robles1999, 2020-09-01:
Just food for thought. If it is a stand alone app, an integrated simple g-code generator like Mash Cam would be the icing on the cake.

mnesarco, 2020-09-01:
It is a standalone app now, but with FreeCAD integration. Everything was rewritten in C++. FreeCAD is an excellent CAD software but as all 3D Cad tools it is very complex for the average Luthier. The main goal of this project is to provide something simple to use with domain specific terminology but powerful enough to model parametric guitar designs without beign a CAD expert.

robles1999, 2020-09-01:
Thats great. You are truly a blessing. I've been having a hard time working with FreeCad. I was using Fusion 360 and got decent at it but don't like the fact that they can stop providing it for free at anytime. In the last 2 months I've been using designspark mechanical and by far is the most user friendly Cad software I've used. It is very powerful and I've been able to create pretty much anything I can think of really fast without any problems. However having your parametric workbench for building guitars is going to make everything so simple and really expand the possibilities.

robles1999, 2020-09-01:
Here are some samples of what I've been able to build on DesignSpark.  These are just shapes I built while learning to use the software, no real guitar dimensions.

[image]
[image]
[image]
[image]
[image]
[image]

mnesarco, 2020-09-20:
I have achieved near perfect surfaces:

[image]

Neck and Fretboard generated in 1.5 seconds.
- Asymmetric neck profiles (double asymmetric)
- Custom asymmetric heel
- Custom headstock
- Custom asymmetric volute
- Custom truss-rod alignment
- Custom headstock angle, neck angle,, zero-fret, scales, radiuses, etc...

robles1999, 2020-09-22:
You are killing me. 8-)  That is crazy good!!!!

mnesarco, 2020-09-22:
> You are killing me. 8-) That is crazy good!!!! 1.5 seconds is amazing. That's some serious code for all the calculations it has to compute. Great job!!

Calculations are the fast part. OCCT boolean operations are the slowest, so I have managed to avoid them as much as possible and I use a highly parallelized code.

PuzzlingMatt, 2020-09-24:
Wow, just wanted to that from reading this thread it looks like this are looking really really good. Look to be able to have a go with the new version soon though I'm having a decent amount of success with the current version. Suspect you might have some more users headed your since it looks like people have started to drop F360 with the recent licence changes.

mnesarco, 2020-09-25:
@PuzzlingMatt yes I knew about the F360 exodus, FreeCAD community is like a refugee land these days. I am working hard to release something as soon as possible. But this is not an easy task. Specially now that I rewrote everything from scratch in C++. But the results are worth the pain.

PuzzlingMatt, 2020-10-19:
@mnesarco anything I can do to help? I can c++ but dont have a lot of experience with freecad - some of its quirks are taking a bit of getting used to! I've actually had some good results with the current version of the workbench (it works fine on windows btw) - just experimenting with the freecad cam stuff now.

mnesarco, 2020-10-20:
Hi @PuzzlingMatt I am having some progress, but there are not deadlines. I do this for fun and love in my spare time. The current dev version is a very different beast compared to the python workbench. This is the parametric double asymmetric neck profile editor:

[image]

By the moment the trussrod channel is aligned with the spine of the profile because I initially thougt that it was the most logic placement. But some more experienced Luthiers think that the trussrod must be aligned with centerline unconditionally, others prefer to decide case by case. So I am adding different alignment options here.

I am also working on more parallelization optimizations, better surfaces, and a very convenient appication file format. I started this as a very simple freecad addon, but it is becoming something more serious now.

robles1999, 2020-10-20:
Loving it.

PuzzlingMatt, 2020-10-20:
Agreed, looks very nice. I guess it'll be something that we might see for sale at some point? Would save me a bucket load of time if I didnt have to model body contours and things by hand since thats always a bit of a faff! Happy to help out with testing or maintaining a windows build if needed in the future.

dantahoua, 2020-11-10:
A bit out of line, but as soon as I manage to compile Freecad 0.19 on my Calculate Linux, count me in the testing guys for your Guitar Design Workbench! It's awesome!!! What Linux distrib. do you use?

mnesarco, 2020-11-10:
Thank you @dantahoua , The version I am working on now does not depend on Freecad, I plan to release it as a standalone AppImage. I am using Mint 19.3 right now.

dantahoua, 2020-11-10:
That sound interesting!!! :)

robles1999, 2021-01-10:
Hello Frank and happy new year!

I was wondering how the project is going. I finally got a CNC and was hoping to design and acoustic guitar. I wanted to check with you before I started designing everything myself which it will take a lot longer than with your software.

mnesarco, 2021-01-12:
Hi @robles1999 , 
The project was paused for some months but I am actively working on it again. I hope I will release an update soon, but I cannot guarantee any specific release date.

robles1999, 2021-01-12:
Hi Frank. Thanks for the update, I was just wondering because I hadn't heard anything from you in a while.

brotherdust, 2021-05-05:
> It is a standalone app now, but with FreeCAD integration. Everything was rewritten in C++. FreeCAD is an excellent CAD software but as all 3D Cad tools it is very complex for the average Luthier. [...]

@mnesarco , you mentioned that it's now written in C++, but didn't mention a link! Please advise where one can get the rewrite! Thank you!

mnesarco, 2021-05-05:
@brotherdust , It is not public yet.

evfool, 2021-06-29:
@mnesarco if you need help developing and would like to make the current state public, I would be interested in also coding, and also using it. Amazing what you have achieved with this project so far.

mnesarco, 2021-06-29:
Hi @evfool thank you for the offer. I am closer to release a new version, here are some pictures:

[image]

[image]

[image]

brotherdust, 2021-08-24:
> Hi @evfool thank you for the offer. I am closer to release a new version, here are some pictures:
> 
> 
> 
> [image]
> 
> 
> 
> [image]
> 
> 
> 
> [image]
> 
> 

Very very cool! I'm so excited for a release. Do you need help with docs? I'm good at that.

brotherdust, 2021-09-13:
@mnesarco , just wanted to check back in with you on your C++ re-write. Was using Guitar Workbench today and ran into some limitations and it seems like your rewrite would fix them. Do you need a tester? I'd like to help somehow, if possible.
Thanks!

mnesarco, 2021-09-13:
Hi @brotherdust I am close but not ready yet to release something:

https://user-images.githubusercontent.com/1087317/133135615-c64072d2-be6f-4b2f-b190-4a421d516af3.mp4

brotherdust, 2021-09-13:
@mnesarco Will it support headless, multiscale builds? 😁

mnesarco, 2021-09-13:
Yes it already support headless and multiscale.

brotherdust, 2021-09-13:
> Yes it already support headless and multiscale.

Oh man! That's incredibly exciting! I'm just a hobbyist but I would seriously pay at least $100 for this software. You got a place I can donate? 😉

mnesarco, 2021-09-13:
I am not receiving donations yet because there is no public product yet. Thank you for your offer. Stay tuned, I want to release the first version this year.

closed 2021-09-23 by mnesarco

mnesarco, 2022-12-12:
Hello @brotherdust ,
The project was seeping for a long time this year, but I have news, I have Gordons Surfaces now thanks to Tigl project.

[image]

It means that finally I can provide transitions for heel-barrel and barrel-headstock with proper continuity. That was the missing piece for an initial release.

brotherdust, 2022-12-13:
> Hello @brotherdust ,
>
> The project was seeping for a long time this year, but I have news, I have Gordons Surfaces now thanks to Tigl project.
>
>
>
> [image]
>
>
>
> It means that finally I can provide [...]

So excited! When?!

mnesarco, 2022-12-13:
I can't compromise a date. Sorry.