mnesarco/MarzWorkbench #8: Bass guitars

TheArtizan, 2021-03-04:
I must say how brilliant your idea is - I would like to adapt to make 4 string through neck bass guitars. There seems to be no way of changing to 4 strings and the neck through it looks like that is the neck joint all the way from top to bottom om body. I must say how brilliant your idea is - I would like to adapt to make 4 string through neck bass guitars. There seems to be no way of changing to 4 strings and the neck through option, looks like that is the neck joint all the way from top to bottom om body. I realise you have created it for 6 string guitars are ther any pointers as to ho I can modify the code to allow 4 string basses, thank you for your time, all the best 25 seconds ago
@TheArtizan

mnesarco, 2021-03-04:
To create a 4 strings neck just set 4 gauges instead of 6:

[image]

Remove last two:

[image]

Then set the string distance at nut and bridge:

[image]

[image]

No code changes required.

TheArtizan, 2021-03-04:
Many thanks that is so helpful
All the best

closed 2021-03-04 by mnesarco