mnesarco/MarzWorkbench #10: Zere Fret

kenf1a, 2021-09-22:
Thanks for making this workbench!  I read that a zero fret neck is supported but I've not been able to find the right parameter or command anywhere within the workbench.  Can you please explain or point me to how this is done?

mnesarco, 2021-09-22:
Simply set the Nut offset to something greater than zero:

[image]

kenf1a, 2021-09-22:
Thanks for the quick reply!  Not sure how I missed that.

One other thing, I'm having trouble getting the nut width correct.  My target is a 42mm nut.  With 36mm string spacing and 3mm margin on each side I should get 42 right?  But now with equal center string spacing I set the distance to 32.5mm and I get a 42mm nut.  Is there something I'm missing?

mnesarco, 2021-09-22:
Maybe you are missing string gauge. You can change string gauges and check again. Also this is measured at zero fret position. So if you are using nut offset greater than zero, the real neck width at nut will be different because the fretboard is not a rectangle.

mnesarco, 2021-09-22:
I did a quick check and apparently side margin is applied twice. I will do a review....

mnesarco, 2021-09-22:
The Side Margin calculation is fixed now. So please upgrade using the Addon Manager or Extman.

kenf1a, 2021-09-22:
Great, thanks for the fast response!  I'm looking forward to your new developments.

mnesarco, 2021-09-22:
[image]

mnesarco, 2021-09-22:
Take into account that the new version is coming so I will soon stop backporting fixes to this version.

https://www.youtube.com/watch?v=qEHmJ4XjxN8

closed 2021-09-22 by mnesarco