mnesarco/MarzWorkbench #11: Fret Nipping or fret width

kenf1a, 2021-09-22:
I've changed the fret nipping from the default of 2mm but I don't see any change in the length of the frets.  This nipping is the distance between the end of the fret and the edge of the fretboard, correct?  I really just want to make one groove across the whole fretboard but I want the fret about 0.5mm short of the edge.

Seems like a bug but let me know.

Thanks!!

mnesarco, 2021-09-23:
Yes, fret nipping is the distance between the fretboard side and the fret slot.
The refresh problem is related to cache. The main reason because I moved from freecad to plain C++ was because of performance. In the freecad version I used a lot of cache techniques combined with a lot of multithreading which is a very complex scenario, so it is not rare to encounter more cache invalidation bugs.

[image]

[image]

kenf1a, 2021-09-23:
I've set fret nipping to 0 but still no change.  I tried rebooting FreeCad but still not fixed.  Is there any quick way to get around this cache bug?

mnesarco, 2021-09-23:
Have you updated to the latest commit?

mnesarco, 2021-09-23:
In case that you cannot upgrade now, just change other parameters to force a cache invalidation.

kenf1a, 2021-09-23:
Yes I updated and was still a problem.  But then I randomly changed the perpendicular fret and then it updated and now the fret width is fixed!  I have a normal fretboard where bass and treble is the same length, so I assume the perpendicular fret does not matter correct?

Thanks again for your very good support!  I look forward to using your new updated tools.  Problem solved.

closed 2021-12-20 by mnesarco