mnesarco/MarzWorkbench #12: Body shape svg file

kenf1a, 2021-09-27:
How does one designate the various holes and pockets in the body shape svg file in Inkscape to correlate with the body top or body bottom or depth, etc...?

mnesarco, 2021-09-28:
Hi, There is a detailed explanation in the wiki page:

https://github.com/mnesarco/MarzWorkbench/wiki/Custom-Shapes

Also you can download the example file and review it:

https://github.com/mnesarco/MarzWorkbench/blob/master/docs/examples/example_custom_body.svg

kenf1a, 2021-09-28:
Thanks again for the quick response.  Here's some comments for you.

1.  "h{p}{num}_{h1}_{h2}" in your wiki looks like a typo. {num} should come before {p}
2. The holes are referenced "t" for top and "b" for bottom.  At first I thought these referred to the body top and bodyback pieces of the guitar, but you are actually referring to the top and bottom of the entire guitar body assembly.  Could be helpful to clarify this.

kenf1a, 2021-09-28:
Actually I think your example has the numbers reversed, not h{p}{num}_{h1}_{h2}.  But check it out.

mnesarco, 2021-09-28:
> 1. "h{p}{num}_{h1}_{h2}" in your wiki looks like a typo. {num} should come before {p}

h{p}{num}\_{h1}\_{h2} is the correct format, but the example in the wiki was reversed. Fixed now.

> 2. The holes are referenced "t" for top and "b" for bottom.  At first I thought these referred to the body top and bodyback pieces of the guitar, but you are actually referring to the top and bottom of the entire guitar body assembly.  Could be helpful to clarify this.

t is for body top
b is for body bottom

closed 2021-09-30 by mnesarco