mnesarco/MarzWorkbench #13: Size of imported SVG files

Cysign, 2021-12-20:
Just checked out your powerful plugin, looks great so far. But I have a strange behavior:
Opening the example files works well.
Ons I edited a single anchor using Illustrator (CS5) or Inkskape (v1.1) and save the file and import it to FreeCad, the size of the whole body is changed, not only of one single curve.

mnesarco, 2021-12-20:
Take care of units. Svg file must be in real size in mm. Units management is not consistent in svg files, but inkscape can manage it. Just be sure that your file's general units are mm.

Cysign, 2021-12-20:
I've uploaded my files here in a zip archive: https://www.file-upload.net/download-14786725/svg_comparison.zip.html

I've checked the measurements using Illustrator and I couldn't find any difference.

mnesarco, 2021-12-20:
I imported one svg into the other one and your drawing is clearly smaller.
[LesStratCaster.svg.zip](https://github.com/mnesarco/MarzWorkbench/files/7742200/LesStratCaster.svg.zip)

mnesarco, 2021-12-20:
Also the viewport of your file is in pixels.

In the provided example file (example_custom_body.svg):
```
   width="601.51996mm"
   height="473.16891mm"
   viewBox="0 0 2273.4613 1788.355
```

In your file: (LesStratCaster.svg)
```
width="1705.096px" 
height="1341.266px" 
viewBox="0 0 1705.096 1341.266" 
```

mnesarco, 2021-12-20:
I don't know about CS5, but in Inkscape you can check:
[image]

Cysign, 2021-12-20:
Wow, thanks!

closed 2021-12-20 by Cysign