mnesarco/MarzWorkbench #14: headstock text in svg file

Cysign, 2021-12-20:
I'm testing the example_custom_headstock.svg right now. I'd like to replace the text 'Test' with another text. But I couldn't figure out how to set it up, so Freecad can read it properly. Is there a documentation about this?
Do I have to use a font and transform it into a vector?

And how would I decide if it's a hole or a pocket? Are those things color coded?

mnesarco, 2021-12-21:
Yes the documentation is in the wiki: https://github.com/mnesarco/MarzWorkbench/wiki/Custom-Shapes

Any shape (including text) must be convertetd to path before importing to freecad.

Cysign, 2021-12-21:
Nice Wiki. But I'm still not surehow to use it.
Can I name the IDs using Inkscape or do I have to modify the file using an text editor?

Cysign, 2021-12-21:
https://www.file-upload.net/download-14788003/screenshot_IDs.png.html

I can see some IDs that look familiar... but all those #pathXXX files?

mnesarco, 2021-12-21:
You can set the id of objects directly in inkscape. I am from my phone now so no snapshot to show you how. But a quick google search can help.

mnesarco, 2021-12-21:
Inkscape / menu / Object / Object Properties

[image]

Cysign, 2021-12-21:
Can I use IDs multiple times or does every shape have to use it'S own ID?

I created a text using the text tool in Inkscape.
Then I transformed this text to path.
Now I gave every object it's own ID, but the result doesn't look what I wanted it to look like:
https://www.file-upload.net/download-14788027/screenshot_headstock.png.html

mnesarco, 2021-12-21:
Once you transformed text to path, there will be many paths, but you can merge all of them into one big path and then assign the id.

There are many Inkscape tutorials: https://inkscape.org/learn/tutorials/ once you know the basic operations it will be very straightforward.

Cysign, 2021-12-21:
I merged the whole path into a single opject. But a single part of it still won't be displayed properly.
Here's my testfile: https://www.file-upload.net/download-14789618/example_custom_headstock.svg.html

I labeled it "lesson 01" - but the 1 won't display proberly.

mnesarco, 2021-12-22:
FreeCAD's svg importer is quite basic. If the paths are not connected, you must set an id to each part.
[example_custom_headstock_fixed.svg.zip](https://github.com/mnesarco/MarzWorkbench/files/7759447/example_custom_headstock_fixed.svg.zip)
[image]

mnesarco, 2021-12-24:
The C++ version (WIP) does not use FreeCAD's svg importer. I wrote an svg importer from scratch highly tuned for this usage. I was trying to release it before xmas, but it was impossible due to time constraints.

closed 2021-12-24 by mnesarco