mnesarco/MarzWorkbench #19: How do I change cavity depths to inches?

TheZakkRoss, 2022-08-03:
I'm just wondering if I can change the depth nomenclature so that it is quicker and is possible more accurate than converting inches to mm! if there is, please let me know because I'm new to cad and svg stuff. This workbench is awesome by the way:)

mnesarco, 2022-08-04:
Hello @TheZakkRoss 
1. I will not support multiple unit systems, I am sorry. Everything is in millimeters in this WB, except for the string gauges.
2. Each fret inlay must be a single svg Path, but the Path can have multiple disjoint parts. So you can create double or multiple part inlays for a fret but you need to combine all of them into a single path (Inkscape: Path -> Combine)

TheZakkRoss, 2022-08-04:
Hey! Thank you for getting back to me so soon! 

1. I remembered I had a caliper that can read in mm so no worries there! 

2. Thank you! I’m glad this work bench has helped me learn so much about Inkscape

mnesarco, 2022-08-04:
Cool.

closed 2022-08-04 by mnesarco