mnesarco/MarzWorkbench #20: Marz Workbench not working on my Mac Os High Sierra

Mojoberlin, 2022-10-21:
**Working on mac os high sierra 10.13.6 with latest FreeCAD Version 0.20-1 and latest Marz Workbench v0.0.26-alpha.** I can launch the Marz Guitar Designer Workbench. I can also open a new instrument. I can see the new instrument in the model view. From here I have several options: create a fretboard or a neck or a body. As an example, if I now create a new fretboard I get the following error message:   

File "/Users/michaelfreudenberg/Library/Application Support/FreeCAD/Mod/Marz/./freecad/marz/command/create_fretboard.py", line 47, in Activated
    App.ActiveDocument.getObject(MarzInstrument_Name).Proxy.createFretboard()
  File "/Users/michaelfreudenberg/Library/Application Support/FreeCAD/Mod/Marz/./freecad/marz/feature/instrument.py", line 164, in createFretboard
    self.add(Fretboard())
  File "/Users/michaelfreudenberg/Library/Application Support/FreeCAD/Mod/Marz/./freecad/marz/feature/instrument.py", line 199, in add
    self.doInTransaction(transaction, f"Marz Add {name}")
  File "/Users/michaelfreudenberg/Library/Application Support/FreeCAD/Mod/Marz/./freecad/marz/feature/instrument.py", line 96, in doInTransaction
    bar = StartProgress(f"Processing {name}...")
  File "/Users/michaelfreudenberg/Library/Application Support/FreeCAD/Mod/Marz/./freecad/marz/extension/ui.py", line 52, in StartProgress
    bar = App.Base.ProgressIndicator()
TypeError: cannot create 'N4Base19ProgressIndicatorPyE' instance

mnesarco, 2022-10-22:
Hi @Mojoberlin I am sorry to hear that it does not work for you. But have you read the mac note in readme? Basically it not supported on mac because I have never tested it on mac.

mnesarco, 2022-10-22:
Looking at your error message it seems like a breaking api change in the progress indicator. I will review tomorrow. Maybe it is affecting all platforms

Mojoberlin, 2022-10-22:
Hi @mnesarco thx for your time and your quick reply, maybe you will find a solution to the problem. If you need any additional information from me then let me know. 👋

mnesarco, 2022-10-23:
There is no problem running it on Linux. So the problem must be specific to Mac or your environment.
[image]

mnesarco, 2022-12-14:
I have added a temp workaround for MacOS.

Referenced FreeCAD bug:
forum: https://forum.freecadweb.org/viewtopic.php?f=10&t=53713
issue: https://github.com/FreeCAD/FreeCAD_Conda/issues/26

mnesarco, 2023-03-25:
As this is apparently solved by now, I am closing this.

closed 2023-03-25 by mnesarco

kellander, 2023-10-27:
Hi!
I have tried all above but can not get a body to load.  I can launch the Marz Guitar Designer Workbench. I can also open a new instrument. I can see the new instrument in the model view.
osx 13.4.1
freecad 0.20.2-29603
marz 00.26 alpha

14:08:02  [MARZ] Traceback (most recent call last):
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Marz/freecad/marz/command/create_body.py", line 48, in Activated
    App.ActiveDocument.getObject(MarzInstrument_Name).Proxy.createBody()
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Marz/freecad/marz/feature/instrument.py", line 176, in createBody
    self.add(Body())
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Marz/freecad/marz/feature/instrument.py", line 199, in add
    self.doInTransaction(transaction, f"Marz Add {name}")
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Marz/freecad/marz/feature/instrument.py", line 96, in doInTransaction
    bar = StartProgress(f"Processing {name}...")
  File "/Applications/FreeCAD.app/Contents/Resources/Mod/Marz/freecad/marz/extension/ui.py", line 52, in StartProgress
    bar = App.Base.ProgressIndicator()
TypeError: cannot create 'N4Base19ProgressIndicatorPyE' instances

Regards!
Jonas

mnesarco, 2023-10-27:
Your version is outdated, please try update to the latest 0.0.27-alpha

kellander, 2023-10-27:
Hi!
Thanks for pointing me in the right direction.

here i find 0.0.25 alpha as "latest".
https://github.com/mnesarco/MarzWorkbench/tree/master

Is there somewhere else I download 0.0.27?

/Jonas

kellander, 2023-10-27:
Found it - add-on manager.
/Jonas

mnesarco, 2023-10-27:
Oh, yes, AddOn Manager should be used always.