mnesarco/MarzWorkbench #23: asymmetric neck shape

zu4lu, 2023-03-10:
Hi Frank, last version of maz is very good, but i'm missing one feature badly, the asymmetric neck shapes(if you ever played one of these, you know what i mean). Is this possible?
Michael

mnesarco, 2023-03-10:
Asymetric necks are supported in the c++ in-development version. But it is not public yet.

mnesarco, 2023-03-10:
https://forum.freecad.org/viewtopic.php?t=58729&start=10#p505612

mnesarco, 2023-04-03:
Thank you. 
The best help I need is in cross platform packaging of the c++ version for distribution.

- Linux: .AppImage  (I can do that)
- Mac: .app (I have no idea)
- Windows: .msi or portable .exe (I have not used Windows for decades... so I have no idea)

I have no timeline, I do this in spare time.

[image]

I have not published it yet because there are many things pending to package this as a product.

mnesarco, 2023-04-21:
@zu4lu 
Yes in the current WB it is possible to define a custom neck profile with a single asymmetry but not compound asymmetry.

https://github.com/mnesarco/MarzWorkbench/wiki/Neck-Profile-Specs

zu4lu, 2023-04-21:
Thank you, will you make it possible it in the workbench?

mnesarco, 2023-04-24:
> Thank you, will you make it possible it in the workbench?

Maybe.

mnesarco, 2024-05-06:
I am working on an editor for neck profiles. It will support double asymmetric necks and it will be available in the FreeCAD version.

zu4lu, 2024-05-06:
Hi, great news, i'm looking forward to it

 
 

"Organization", "name": "GitHub", "url": "https://github.com" } } ]***>
lg
Michael Kaufmann

zu4lu, 2024-06-09:
Hi Frank, maybe this will help. I make my shapes based on this and its fantastic.
https://www.kappi.com/blog/2012/10/asymmetric-neck-shapes/

mnesarco, 2024-06-09:
@zu4lu thank you for the link, but i already have developed the double asymmetric neck editor in my private c++ version.  I just have had not time to port it here. The idea is actually pretty similar:

[image]

zu4lu, 2024-06-09:
good news, thanks

mnesarco, 2025-07-15:
> Oh wow, would it be possible to desig an own shape like an enduro neck?

Not currently. There are infinite variations for each aspect of a guitar, it is very hard to cover all possible cases. Said that, it is not that difficult to make the custom neck profile more flexible, but it also makes the UI more complex. The idea of this Workbench is to cover common cases for amateur guitar makers and for experimentation. Automation will never cover all the Artistic or advanced features required by professionals.

Cysign, 2025-07-15:
That makes sense. I'd still like to build my first homemade guitar with an Enduro neck (since I can't afford a Boden Prog NX 7 - just to see if I like it :D ). So I have to get as close as possible to that design and to the rest by hand. Thanks a lot for all your great work in this project <3