mnesarco/MarzWorkbench #28: Custom Neck Pockets

Arayl96, 2024-02-29:
Hello as stated in my YouTube comment I thank you again for your help so far, so creating a way to edit the size of the neck pocket other than length and depth. I’m not sure how you can do it but I was thinking of you possibly modifying your Les Paul body svg and creating an existing path or something like the pickups or cavities, something so people can just download the svg and modify the svg like you can currently do in Inkscape, so pretty much the Les Paul downloadable svg with a neck pocket with a pre existing path that is customizable with the nodes in Inkscape, that will be transferred to freecad like all other parts of the body. I’m not sure if that is a way to do it but you are the brains and know this better than anyone. Just something so people with a custom neck or an already existing neck can fit to the guitar body so they can make sure it’s a perfect fit and not worry about the automatic neck pocket being to wide or the automatic taper of the pocket being to great.

mnesarco, 2024-02-29:
Currently you can do your own pocket in the body svg file, it will be managed like any other pocket. The only problem is that the automatic neck pocket will interfere, so the first thing I can do is making the automatic neck pocket optional but set by default to not break existing models. I will add a parameter so the user can activate/deactivate it.

Arayl96, 2024-02-29:
We can see how that works, and creating a pocket on the svg is possible it will require some extra work on my part to make sure the pocket is perfectly centered, which isn’t really a problem. I don’t wanna make it sound like I’m rushing because I know this takes a lot of time and you won’t just push it for one person, but is there a time frame this would be implemented or the update containing it would be released, I’m not in a hurry or anything I just wanna make sure I’m one of the first to use it, again thank you for the help and I am really thankful that you created this whole process and program it makes everything so easy, I know it took a lot of work and probably wasn’t easy for you to create but I really appreciate it an I know many others do as well

mnesarco, 2024-03-01:
I have made the automatic pocket optional. You can upgrade and test it.

mnesarco, 2024-05-06:
Generating custom arbitrary neck pocket from svg and matching with generated neck is too complex. I am closing this as wont fix.

closed 2024-05-06 by mnesarco