mnesarco/MarzWorkbench #29: Body Carves

Arayl96, 2024-03-11:
I know you mentioned it I believe in one of you videos but I would like to request a way to add top and body carves if it is possible

mnesarco, 2024-03-11:
do you mean bevels and belly cuts, or something more complex like a LP top carving?

Arayl96, 2024-03-12:
Both would be good, I know it is really complex and I’m sure complicated so it’s fine if it can’t be done

mnesarco, 2024-03-12:
It is not implemented in the current version. But armrest and belly cut are definitively less complex than top carving.

mnesarco, 2024-03-25:
I suppose the binding channel can be done with the OCCT Offset functionality, but it is a bit unpredictable sometimes. I have to check

Arayl96, 2024-03-25:
I think it would be a nice addition to the software, if you find out the best way to do it or if you are thinking of implementing it, feel free to let me know

mnesarco, 2024-05-06:
Body carves (Ergonomic cutaways) are available now (0.1.3)

mnesarco, 2024-05-06:
The title of this issue is too ambiguous, it can refer to carved tops or ergonomic cutaways, and also you mention bindings.
So as ergonomic cutaways are supported now, i am closing this issue as completed. For carved tops and bindings you can add new issues. Please one specific thing per issue.

closed 2024-05-06 by mnesarco