mnesarco/MarzWorkbench #33: Not an issue but i need some help

maximizer12, 2024-05-06:
after downloading the workbench from the addon manager the report view window says the following: 
(i already downloaded Pyside and GitPython and the required curves workbench so idk what is going on.)

23:43:47  During initialization the error "No module named 'PySide.QtWidgets'" occurred in freecad.marz
23:43:47  --------------------------------------------------------------------------------
23:43:47  Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/maxi/.FreeCAD/Mod/Marz/freecad/marz/init_gui.py", line 23, in <module>
    from freecad.marz.extension.threading import task, Task
  File "/home/maxi/.local/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/maxi/.FreeCAD/Mod/Marz/freecad/marz/extension/threading.py", line 24, in <module>
    from freecad.marz.extension.qt import QtCore
  File "/home/maxi/.local/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/maxi/.FreeCAD/Mod/Marz/freecad/marz/extension/qt.py", line 47, in <module>
    from PySide.QtWidgets import (  # type: ignore
  File "/home/maxi/.local/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'PySide.QtWidgets'
23:43:47  --------------------------------------------------------------------------------

i will be very greatful if someone knows what else should i download.

mnesarco, 2024-05-06:
Are you using an old freecad version. It requires 0.21.1+

maximizer12, 2024-05-06:
thank you for pointing that out again but somehow I cant update from 0.19 to 0.21.1 I reinstalled it and it still downloads the old version.

mnesarco, 2024-05-06:
@maximizer12  https://github.com/FreeCAD/FreeCAD/releases/tag/0.21.2

maximizer12, 2024-05-07:
thank you a lot for the help i got it working

closed 2024-05-07 by maximizer12