mnesarco/MarzWorkbench #34: Truss rod channel

mpd75, 2024-05-07:
Not sure if it is something design specific or something in 0.1.4 - I've used the standard truss rod channel settings (haven't changed them) but the truss rod channel is ending up proud of the neck... removing the fretboard shows it as being raised from the neck rather than routed into the neck...
[image]

mnesarco, 2024-05-07:
The image is not clear. I will check tonight but it is better if you share your FCStd file.

mpd75, 2024-05-07:
Sorry - will close issue... what I've done is open one of your example files, change the custom body shape, then change the parameters, update, change neck, update, change headstock... and so on.  So I end up with my design and working perfectly.  Will look again sometime at order of events and see if I can figure out what caused this, but I have a way of working with it.