mnesarco/MarzWorkbench #37: Custom neck pocket and Input problem

Arayl96, 2024-06-02:
Hello, am having problems with custom neck pockets. i am unable to fine the option to turn off the automatic neck pocket. So i have the custom neck on the body but it still add the auto neck pocket, also freeCAD is not wanting to accept the inputs Ive made as far as the neck depth and length goes. As you can see in the screenshot the neck pcoket depth and length are not reflecting on the 3D body

[image]

[image]

Arayl96, 2024-06-02:
Here is a screenshot of my inkscape, also wholes aren't carrying over, all i did was duplicate existing paths and changed the label which worked in the past but isn't working now. As far as the neck pocket goes it did work before but i tried to open a previous file where the neck pocket was right and the auto neck wasnt applied and it add the auto neck pocket to the file when i just tried to open it in freeCAD

[image]

Arayl96, 2024-06-02:
nothing changed between the to because i iopen the double cut file and moved nodes to make the single cut all i did was delete the f hole and extra pickup and move and duplicate the holes

mnesarco, 2024-06-02:
You mean that this does not work?
[image]

Depth is still needed to position the neck.

Arayl96, 2024-06-02:
I am unable to find that, how do you get to that access that panel?

mnesarco, 2024-06-02:
@Arayl96 you need the latest version. You have a pretty outdated version installed.

Arayl96, 2024-06-02:
Okay I was about to update it before I opened this thread, I’ll update it and let you know if it works or if I have any problems

Arayl96, 2024-06-03:
I got it to work, is there a way you can add a way to change the width of the neck pocket generated? the current pocket that generates with the neck is about 61mm. Say i wanted to delete the neck and fret board that are auto generated so it keeps the pocket but i have a neck with a straight neck heel of 56mm. and then I have big neck i wanna use like 67mm neck heel, could you add something to where that is adjustable? I added a picture to show you the neck pocket width and to be clear about the measurement I'm talking about, becuase when i disable neck pocket the pocket i have on my svg doesnt generate when it use to and you can see it use before i updated as seen in the original pictures(i added one again so you can see the neck for the svg was being generated). I didn't changing anything in the svg just updated the workshop and freeCAD. Is it also possible to add a option to add a zero fret to the neck, a zero fret is an extra fret near the nut used to keep the stings at proper height and spacing, I included a picture of one. If you are able to add the neck pocket width feature and the zero fret I could make my own neck through the workshop and would have no need to try and fit my custom neck file.

[image]

[image]

[image]

mnesarco, 2024-06-03:
I don't now if you want the neck pocked to be generated or not. You asked for how to disable the generated neck pocket and now you ask how to adjust the generated neck pocket.

Anyway, the generated neck pocket depends on the bridge reference, pocket depth and on the neck dimensions which are generated from nut width and string distance at bridge and Fretboard margins. There is no way to specify arbitrary dimensions like neck pocket width at specific position. I cannot account for all the infinite possibilities.

About the zero fret, you can just specify the nut offset at zero fret:

[image]
[image]

I know that sometimes it is frustrating, but guitar design is an art, and it is impossible to cover all the artistic possibilities in an automated way. The last resort is to model the instrument using the FreeCAD modeling tools directly from scratch or to generate some part of the work and continue modeling by hand.

mnesarco, 2024-06-03:
I also suggest you to add a proper bridge reference to the correct pocket alignment:

https://github.com/mnesarco/MarzWorkbench/wiki/Custom-Shapes

[image]

Arayl96, 2024-06-03:
Okay great, I didn’t know you updated the template I’ll definitely update what I currently have, one final question, when I put my exported stl file into a software for routing/printing these blocks/plugs appear, they don’t show up in freeCAD only softwares like easel I added a screen shot
![IMG_6219]
(https://github.com/mnesarco/MarzWorkbench/assets/161767535/36d894d5-18f9-4a4a-bbde-7ac8ad9bde85)

mnesarco, 2024-06-03:
@Arayl96 those are not generated by the wb. Maybe you already have them in the slicer or in FreeCAD when exported.

Arayl96, 2024-06-03:
Okay ill try to look for the in freeCAD

closed 2024-06-05 by mnesarco