mnesarco/MarzWorkbench #38: Neck Pocket Into Slicing Software

Arayl96, 2024-06-05:
Hey i tried to put a body and template into a slicing/CAM software and the top and bottom layers are separating, the only change i made to the file from the previous body in the screenshots before was disable the neck pocket, when it was previously enabled. And since it is still considered the same file i cant edit it in the CAM software. 

[image]

[image]

[image]

[image]

mnesarco, 2024-06-05:
Body Top and Body Back are separated pieces. What is the problem?
Do you want a single piece body? in that case set top body thickness to zero.

Arayl96, 2024-06-05:
I know they are separate pieces but I don't want them separated I want them together like in the first two screenshots. If I set the top thickness to zero than I wont get the holes for the pots and the cavity will be open. like in the third screenshot. They aren't separated in a CAM software when a neck pocket is generated but they are separated in the software when a neck pocket isn't generated. And i'm unable to edit the CAM software to close the gap of the separation.

[image]

[image]

 [image]

mnesarco, 2024-06-05:
The Marz Workbench generate two Parts , one for the Top and one for the Back in this scenario. Even that you see all the parts "assembled" in FreeCAD it does not mean they are prepared for a specific CAM workflow.  How you position or manage them in your **external software** (slicer/CAM) is out of scope. From the CAD perspective, Top and Back Parts are separated solids, you can fuse them if it is what you want, but there is a high probability that you get an object impossible to produce in a CNC router. Take into account that CNC routing is a subtractive process while 3D printing is an additive process. I don't mean that you cannot do that, but just be sure that you understand the process. CAM is not trivial and unfortunately i cannot help you with your external software, but maybe _inventables/easel_ forums can help.

mnesarco, 2024-06-05:
@Arayl96 I understand your frustration, but as i said, I can't help you with the external software. On the CAD side, the Parts are correctly generated, placed and exported as individual parts. I barely have time to support this project and i see no bug here. Just fuse the two Parts before export as what you want is to machine both at once.

mnesarco, 2024-06-05:
I would be great if some user with inventables/easel experience contribute some comments here to help Arayl96.

mnesarco, 2024-06-07:
Hello @Arayl96 I am closing this issue as "out of scope". I'm sorry I couldn't help you this time.

closed 2024-06-07 by mnesarco