mnesarco/MarzWorkbench #39: instablity inside FreeCad

0GURUGUAY, 2024-08-21:
SO I use the workshop from 2 days. Everything great but my mayor concern is about that at the first time, the window to configure the guitare is opening. No problem. I do my config, working great. But if I go back to my guitare, I am loosing access to the wworkshop windows and there is no way to get it back. Only to close my doc and open it again...
Do I miss something ?
Gracias
Max

0GURUGUAY, 2024-08-21:
using FC 0.21.2-33771

mnesarco, 2024-08-21:
Do you mean that this icon does nothing after the first time?

[image]

1. The workbench must be activated from the workbenches selector or tab depending on your config
2. When the Workbench is active, the toolbar is visible, so you just click the icon

0GURUGUAY, 2024-08-21:
Yes, the workbench is active.

With the doc opened, the first click on the image is OK, opening the window.
But if the window is closed, impossible to open it again.
I have to close the doc and open it again.
I have rebooted my MacBook Pro M1 - 16Gb to be sure.
And no error message
--------------
few second laster......
--------------
the window was behind Fcad....
Right now it is working....

Better !
thank you

Max

mnesarco, 2024-08-21:
Ok, it sounds like an Issue related to Qt/PySide on Mac. It is not rare that windows behave differently on mac. I will check, maybe i have to specify the window z-order explicitly for Mac env.

mnesarco, 2024-08-22:
@0GURUGUAY  I have added an extra trick to force raise the window to front on MacOS, can you test the latest version in your environment?

0GURUGUAY, 2024-08-22:
done ! 1.7 installed. Looks OK - if something goes wrong I will alert you. Thank you

closed 2024-08-22 by mnesarco