mnesarco/MarzWorkbench #40: head problem

0GURUGUAY, 2024-08-22:
I am drawing this telecaster head and get trouble to get it working.
I try your head and it is fine, so my problem is inside my drawing.
I put the contour, etc.... but no way. You can see it in red with Update draft. 
Where should I look for ? 
thank you.
Max

0GURUGUAY, 2024-08-22:
my svg
[image]

0GURUGUAY, 2024-08-22:
my last try.... I tested a lot of thing but get every time this cut....
[image]

mnesarco, 2024-08-22:
Unfortunately, the algorithm used to create the smooth transition does not work equally well with all possible head stock shapes. Some minimum distance is required from transition and nut, but it depends on the shape, it is a bit fussy.

First, your contour path is not closed, it must be a closed path:
[image]

Secord, for some reason your reference lines are actually curves, it should not be a problem but i replaced them for actual lines.
[image]

Telecaster headstock is a very problematic shape on the bottom side because it is too straight, the algorithm expect something a bit more distant from the center, it could generate an ugly connection surface there if the is no bump.
[image]

Finally, the distance between the transition and the nut must be a bit larger, you have to experiment until yo get it working without being to difference from your original shape.

[image]

This is an approximation:
[image]

And finally, the "fixed" svg file:

[image]

mnesarco, 2024-08-22:
I have to improve the algorithm to reduce the required distance from transition to nut.

mnesarco, 2024-08-22:
I have added a small adjustment in 0.1.8, so your original file works now just closing the open path.

[image]

0GURUGUAY, 2024-08-22:
update, working well, thank you, I appreciate your help.

mnesarco, 2024-08-22:
Great!!

closed 2024-08-22 by mnesarco