mnesarco/MarzWorkbench #43: Please update Inkscape extension for Inkscape 1.4

levito, 2024-11-11:
Hi Frank, I'm currently designing a custom bass in Inkscape and wanted to use your Marz Workbench to convert it to 3d. I recently updated Inkscape to version 1.4 and I realized that I can't install the Marz extension. It's only compatible with Inkscape 1.3.2.

Could you please update it to work with Inkscape 1.4? Is it just a version number change, or do you expect other things to require adjustments?

Thanks for your awesome work, I'm really looking forward to make use of it and contribute wherever I can!

mnesarco, 2024-11-11:
Hello @levito 

I added the 1.4 tag in the inkscape extension. but for some reason it is not listed in the ExtensionMananager (which is very buggy)

One quick solution is to just download the zip from the page and unzip it into your extensions folder.

https://inkscape.org/~mnesarco/%E2%98%85marz

levito, 2024-11-12:
Awesome, thank you, I'll give it a try!

levito, 2024-11-12:
Looks like it works just fine in Inkscape 1.4. I will close this issue since it's not a problem with the extension itself.

closed 2024-11-12 by levito