mnesarco/MarzWorkbench #44: Straight headstock: zero volute radius doesn't remove it

levito, 2024-11-15:
When setting the headstock angle straight, using 0 as volute radius should make it straight. Instead, it wil create a large bump, see screenshot:

[image]

As a workaround, I entered a very large value, like 10000, as a radius. However, this is not really intuitive:

[image]

mnesarco, 2024-11-21:
@levito can you please share your file?

levito, 2024-11-21:
Sure, here you go:

[image]

And here is what I used as parameters:

[image]
[image]

mnesarco, 2024-11-21:
Thank you. confirmed.