mnesarco/MarzWorkbench #45: Reopened files can't be edited with Guitar tools

levito, 2024-11-15:
When keeping a newly created file open, adjustments can be made, both in measurements and via changing the SVG files. This works both via the pick icon and via double-click on "Instrument Parameters" in the tree view.

However, closing and reopening a file makes this impossible. I get the following error:

[image]

I could reproduce this in FreeCAD 1.0 RC2, FreeCAD 1.0 RC4, and the most recent weekly build of Ondsel ES:

OS: macOS 14.7
Word size of Ondsel: 64-bit
Version: 2024.3.0dev.38472 (Git)
Build type: Release
Branch: flavor
Hash: 60c383a71ad9a470a811ccf6537156e03f180539
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * Ondsel-Lens 2024.10.21.01
  * Marz 0.1.11

mnesarco, 2024-11-16:
Thank you for reporting. Fixed in 0.1.12

closed 2024-11-16 by mnesarco

levito, 2024-11-16:
Thank you, this fixed the issue, even for files saved with version 0.1.11!

commit f5b30262 (2024-11-16): Fix #45: Invalid count of widgets