mnesarco/MarzWorkbench #50: importSVG overhaul breaks MarzWorkbench

julian7, 2025-05-14:
I'm running weekly builds, and recently I found SVG import is not working properly. It's very likely that FreeCAD/FreeCAD#20749 caused this.

mnesarco, 2025-05-15:
@julian7 Thank you for the report. Can you provide more information?

- FreeCAD Version
- MarzWorkbench version
- Python version

What error is reported if any or what is the affected behavior? Example svg file?

julian7, 2025-05-15:
My version:

```
OS: macOS 15.4.1
Architecture: arm64
Version: 1.1.0dev.41464 (Git)
Build date: 2025/04/28 16:20:16
Build type: Release
Branch: main
Hash: 8d6547d519ffb331cfba054b4db2750b3e71339b
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 
shiboken , xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/Hungary (en_HU)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * btl 0.9.9
  * FeedsAndSpeeds 0.6.0
  * CamScripts
  * Marz 0.1.15
```

For a test, I've exported the Red Special's body shape from the example, and re-imported it. It gave the following popup:

[image]

The addon's log shows the following:

```
Importing /Users/js/src/private/red-special-body.svg
Validating /Users/js/src/private/red-special-body.svg
contour: Required. Not found
```

At the same time, this is the whole contents of the Report view:

```
08:36:29  (qt.qpa.fonts) Populating font family aliases took 193 ms. Replace uses of missing font family "Courier" with one that exists to avoid this cost. 
07:38:21  processing element 1: svg
07:38:21  existing group transform: []
07:38:21  existing group style: []
07:38:21  name: svg1
07:38:21  done processing element 1
07:38:21  processing element 2: sodipodi:namedview
07:38:21  existing group transform: [Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1))]
07:38:21  existing group style: []
07:38:21  name: namedview1
07:38:21  done processing element 2
07:38:21  processing element 3: defs
07:38:21  existing group transform: [Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1))]
07:38:21  existing group style: []
07:38:21  name: defs1
07:38:21  done processing element 3
07:38:21  processing element 4: g
07:38:21  existing group transform: [Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1))]
07:38:21  existing group style: []
07:38:21  name: layer1
07:38:21  done processing element 4
07:38:21  processing element 5: rect
07:38:21  existing group transform: [Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1)), Matrix ((1,0,0,112.713),(0,1,0,-11.1125),(0,0,1,0),(0,0,0,1))]
07:38:21  existing group style: [[None, (0.09803921568627451, 0.09803921568627451, 0.09803921568627451, 0.0), 2.0]]
07:38:21  name: rect6
07:38:21  done processing element 5
07:38:21  processing element 6: path
07:38:21  existing group transform: [Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1)), Matrix ((1,0,0,112.713),(0,1,0,-11.1125),(0,0,1,0),(0,0,0,1))]
07:38:21  existing group style: [[None, (0.09803921568627451, 0.09803921568627451, 0.09803921568627451, 0.0), 2.0]]
07:38:21  name: h42_0_2635
07:38:21  data: {'id': ['h42_0_2635'], 'clip-path': ['none'], 'style': 'fill:none;stroke:#008000;stroke-width:1.0;stroke-linecap:round;stroke-linejoin:round;stroke-miterlimit:10;stroke-dasharray:none;stroke-opacity:1;vector-effect:non-scaling-stroke;-inkscape-stroke:hairline', 'd': ['M', '162.486', '169.331', 'h', '-6.32494', 'v', '-3.8251', 'c', '0', '-1.75359', '-1.42169', '-3.17493', '-3.175', '-3.17493', 'h', '-8.65008', 'c', '-1.75327', '0', '-3.17497', '1.42134', '-3.17497', '3.175', 'v', '3.8251', 'l', '-6.32498', '-7e-05', 'c', '-1.75334', '0', '-3.17504', '1.42134', '-3.17504', '3.175', 'v', '64.65', 'c', '0', '1.75363', '1.4217', '3.17497', '3.175', '3.17497', 'h', '6.32499', 'l', '3e-05', '3.82509', 'c', '0', '1.75357', '1.4217', '3.17491', '3.175', '3.17491', 'h', '8.65008', 'c', '1.75328', '0', '3.17497', '-1.42134', '3.17497', '-3.175', 'v', '-3.82508', 'l', '6.32495', '8e-05', 'c', '1.75365', '0', '3.17499', '-1.42134', '3.17503', '-3.175', 'v', '-64.65', 'c', '-4e-05', '-1.75363', '-1.42138', '-3.17497', '-3.17504', '-3.17497', 'z'], 'sodipodi:nodetypes': ['cccccccccccccccccsccc'], 'inkscape:label': ['h42_0_2635'], 'fill': 'none', 'stroke': '#008000', 'stroke-width': '1.0', 'stroke-linecap': 'round', 'stroke-linejoin': 'round', 'stroke-miterlimit': '10', 'stroke-dasharray': 'none', 'stroke-opacity': '1', 'vector-effect': 'non-scaling-stroke', '-inkscape-stroke': 'hairline'}
07:38:21  Adding wire for 'h42_0_2635' - reason:  no-fill.
07:38:21  done processing element 6
[... 547 lines cut ...]
07:38:21  existing group transform: [Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1)), Matrix ((1,0,0,112.713),(0,1,0,-11.1125),(0,0,1,0),(0,0,0,1))]
07:38:21  existing group style: [[None, (0.09803921568627451, 0.09803921568627451, 0.09803921568627451, 0.0), 2.0]]
07:38:21  name: h56_0_2635
07:38:21  data: {'style': 'vector-effect:non-scaling-stroke;fill:none;stroke:#008000;stroke-width:1.0;stroke-linecap:round;stroke-linejoin:round;-inkscape-stroke:hairline;stroke-dasharray:none', 'd': ['M', '12.6437', '283.753', 'L', '35.6733', '328.584', 'L', '92.2387', '340.526', 'L', '67.6799', '296.784', 'Z'], 'id': ['h56_0_2635'], 'inkscape:label': ['h56_0_2635'], 'vector-effect': 'non-scaling-stroke', 'fill': 'none', 'stroke': '#008000', 'stroke-width': '1.0', 'stroke-linecap': 'round', 'stroke-linejoin': 'round', '-inkscape-stroke': 'hairline', 'stroke-dasharray': 'none'}
07:38:21  Adding wire for 'h56_0_2635' - reason:  no-fill.
07:38:21  done processing element 85
07:38:21  processing element 86: path
07:38:21  existing group transform: [Matrix ((1,0,0,0),(0,1,0,0),(0,0,1,0),(0,0,0,1)), Matrix ((1,0,0,112.713),(0,1,0,-11.1125),(0,0,1,0),(0,0,0,1))]
07:38:21  existing group style: [[None, (0.09803921568627451, 0.09803921568627451, 0.09803921568627451, 0.0), 2.0]]
07:38:21  name: midline
[... cut ...]
07:38:21  data: {'style': 'vector-effect:non-scaling-stroke;fill:none;stroke:#008000;stroke-width:1.0;stroke-linecap:round;stroke-linejoin:round;-inkscape-stroke:hairline;stroke-dasharray:none', 'd': ['m', '106.07337', '168.0729', '1.5875', '73.51585'], 'id': ['bridge'], 'inkscape:label': ['bridge'], 'vector-effect': 'non-scaling-stroke', 'fill': 'none', 'stroke': '#008000', 'stroke-width': '1.0', 'stroke-linecap': 'round', 'stroke-linejoin': 'round', '-inkscape-stroke': 'hairline', 'stroke-dasharray': 'none'}
07:38:21  Adding wire for 'bridge' - reason:  no-fill open Wire.
07:38:21  done processing element 87
07:38:21  closing group
07:38:21  closing group
```

I hope it helps.

PR #51 (2025-05-19): allow "_w" or "_w000" ending of path IDs
Closes #50

Apparently, Hungarian Notation got its foot into ImportSVG's door. With the recent Import SVG changes, all wires (eg. all shapes with no fill or no closing) get a "_w" suffix (see FreeCAD/FreeCAD@5f8eac49f31626354ee69bf40616bac801ca5560, https://github.com/FreeCAD/FreeCAD/blob/5f8eac49f31626354ee69bf40616bac801ca5560/src/Mod/Draft/SVGPath.py#L768).

Supporting Number of copies suffix might be not required (as they appear only repeated imports), but I certainly had them when I tested it multiple times in python console.
closed 2025-05-19 by mnesarco