mnesarco/MarzWorkbench #57: Neck Pocket Lenghth or Manual position

shIxx01, 2025-08-15:
Hello Frank,

I've been trying to do some exercises with your fantastic Marz WB and have been trying to follow this tutorial. I'm encountering a problem at minute 13:50:
[Link](https://youtu.be/XxVnvgsdRSI?t=831)

In my case, it looks exactly like that, with the body offset. Now, in the video, you adjust "Neck Pocket Length." In the new preference window (since 0.1.0), the entry is called "Manual Position." But it doesn't matter what value I enter, the pocket length doesn't change. 

Is this a bug, or did I misunderstand where the bridge line should be drawn?

[image]

[image]

mnesarco, 2025-08-15:
Manual position only works if you don't have a Bridge reference defined in your svg file. IF you have the bridge reference, it has absolute precedence and manual position is ignored.

shIxx01, 2025-08-15:
Oh, I see, that makes sense. Thanks for the quick reply. The guitar is a template from the site you recommended in the video, so credit goes to [electricherald](https://www.electricherald.com/music-man-albert-lee/)